LowellInstruments/Domino #34: Converting to current without magnetometer/accelerometer enabled is confusing

Nickel33, 2018-09-23:
STORY
As a user, I selected an output type of "Current" by accident on a file that didn't have the magnetometer and accelerometer channels enabled.  The file did not convert but I didn't notice right away because other files in the list did convert.  When I went and looked for the temperature output file from the file without the accl and mag enabled, I didn't see it and was confused.  I tried converting again and eventually noticed the "X" next to the file and wondered what that was.  Without specific feedback, I was confused.

Nickel33, 2018-12-08:
There needs to be some indication that a file that was converted does not support the conversion type. I.e. converting to current without the accelerometer and/or magnetometer enabled.

jeff101, 2019-09-24:
This was resolved in 0.6.2

Nickel33, 2019-10-23:
The "Error" status makes it clear that something went wrong.  But it is not very informative.  Closing.

closed 2019-10-23 by Nickel33