LowellInstruments/Domino #51: Make sensor_refresher more robust

mo-nathan, 2018-10-23:
STORY:
As a user
I don't want the GUI to spontaneously fail due to unexpected output from the Logger
so I have more confidence in the software and Lowell Instruments in general

DONE:
The GUI can run for 24 hours with no failures.

BACKSTORY:

The UI failed for me today with the following two stack traces:

nathan@nathan-XPS-13-9370:~/lowell-instruments/src/Logger$ make run
Traceback (most recent call last):
  File "/home/nathan/lowell-instruments/src/Logger/gui/sensor_refresher.py", line 27, in refresh
    self._set_item_text(index, 1, enabled_string(sensor, readings))
  File "/home/nathan/lowell-instruments/src/Logger/gui/sensor_refresher.py", line 37, in enabled_string
    if sensor in readings:
TypeError: argument of type 'NoneType' is not iterable
Aborted (core dumped)

and

nathan@nathan-XPS-13-9370:~/lowell-instruments/src/Logger$ make run
Traceback (most recent call last):
  File "/home/nathan/lowell-instruments/src/Logger/gui/sensor_refresher.py", line 23, in refresh
    readings = self.logger_controller.get_sensor_readings()
  File "/home/nathan/lowell-instruments/src/Logger/venv/lib/python3.6/site-packages/mat/logger_controller.py", line 226, in get_sensor_readings
    return SensorParser(sensor_string, self.converter).sensors()
  File "/home/nathan/lowell-instruments/src/Logger/venv/lib/python3.6/site-packages/mat/sensor_parser.py", line 54, in sensors
    self.add_raw_sensors(count)
  File "/home/nathan/lowell-instruments/src/Logger/venv/lib/python3.6/site-packages/mat/sensor_parser.py", line 64, in add_raw_sensors
    self._sensors[name] = self._sensor_value(i, name)
  File "/home/nathan/lowell-instruments/src/Logger/venv/lib/python3.6/site-packages/mat/sensor_parser.py", line 69, in _sensor_value
    index not in [0, 8])
  File "/home/nathan/lowell-instruments/src/Logger/venv/lib/python3.6/site-packages/mat/utils.py", line 52, in four_byte_int
    result = int(bytes[2:4] + bytes[0:2], 16)
ValueError: invalid literal for int() with base 16: '0\r00'
Aborted (core dumped)

mo-nathan, 2018-10-24:
Wrote tests that hit the above errors and then fixed them.
Still need to run the longer test.

Nickel33, 2018-12-08:
Nick to attempt a 24 hour connected to status window test.

Nickel33, 2019-01-03:
18 hours and going strong.  Closing.

closed 2019-01-03 by Nickel33