LowellInstruments/Domino #72: Real-time Data box should be labeled, needs yand should be vertically scaled.

Nickel33, 2018-12-05:
On the Start/Stop tab, the real-time data box needs a few changes:
1. The units are missing.
2. The Enabled column is too wide.
3. The table should be labeled "Real-Time Data" when the logger is stopped and "Most-Recent Data" when it is not.
4. The table should auto-size in the vertical dimension for the number of rows.  (i.e. no blank white space under the "Battery" Row.)
4. When using a logger with firmware that does not support the GLS command, the "Enabled" Column should be either removed or listed with a dash (-).

Nickel33, 2018-12-09:
Phase 1:  If GLS returned is valid, then show enabled/not enabled (yes/no).  If GLS not supported, show dash.

Phase 2 for future:  If GLS not enabled, show dash until values change, then switch to yes.

jeff101, 2019-10-01:
6. Center justify the columns

Nickel33, 2019-10-23:
Nit picking here.  Two requests that I think will make the tables more consistent and easier to read:
1. Change the column headings to bold so that they match the same style as the column headings on the "Convert Files" screen.
2. Shade the top row (with the headings) on both the "real-time" and "convert" tables to differentiate the heading row from the rest of the table.

jeff101, 2020-01-08:
It turns out that Windows 10 has done away with the horizontal border between header and table. For example, this is Windows 10 explorer:
[image]

However, it looks stupid when there are grid lines below. I added a provision to the code that will add the header border for Win 10 only.

Nickel33, 2020-02-06:
Looking good version 0.6.4.  Closing.

closed 2020-02-06 by Nickel33