LowellInstruments/Domino #74: Add ability to split large .lid files

Nickel33, 2018-12-06:
STORY:
I am a personal computer and have only limited memory.  I can't open a 2-3 GB .lid file  nor can I manage a 10-20 GB .txt file.  I need a way to split the file into easily manageable chunks when parsing it.

Details:
At a minimum, we need support that is the same as MLC.

jeff101, 2018-12-19:
This has been implemented in the back-end. Now we just need to decide where to put the option in the GUI. Adding it to the "file output options" dialog would seem to be the logical place. At the bottom of  "general" tab? Or maybe on it's own tab?

Nickel33, 2018-12-21:
I think it should go user "File Output Options".

Nickel33, 2018-12-21:
Per our conversation on 12/21:
Put "Split file" in a new tab under options.
Move Average Burst to the new tab.
Move Output format to the new tab.
Move declination adjustment to the main window (replacing output format).

Nickel33, 2019-07-25:
Verified fixed 0.5.8.

closed 2019-07-25 by Nickel33