LowellInstruments/Domino #76: Time/Clock Issues

Nickel33, 2018-12-06:
STORY:
As a user, I want to be sure that I start the logger with the correct time.

Details:
There are several issues:
1. The software is not highlighting the logger clock in yellow when it is more than 1 minute out of sync.
2. The software is not warning the user with a pop-up window when the PC and logger clock are out is sync by more than 1 minute.
3. The software is not adjusting the "computer time" field when the date and time are adjusted by the user in the control panel. Rather, the user must close and re-open the application.  This is important because many customers set their logger to UTC time but normally have their PC on local time.  So without this change, they need to set the PC time zone before opening the application.  That is unusual workflow.

Nickel33, 2018-12-20:
No fixed for the case where the user adjusts the clock while Domino is open. To reproduce the error:
1. Set your loggers clock to the correct time
2. Close Domino.
3. Manually adjust the PC clock to be 2 minutes off.
4. Open Domino and observe that the time is highlighted.
5. Set the logger's clock (highlighting disappears).
6. Open PC clock settings and set clock back to correct time.
7. Observe that the logger time is NOT highlighted and the logger can be started with no warning.
8. Wait a few minutes and eventually the logger time will be highlighted, but by that time it is too late.

The fix is to compare the logger and PC time much more frequently so that the logger can't be started without a time check first.

Nickel33, 2019-01-03:
Check this issue on the Mac to be sure the logger time is flagged if the clock is out by more than 1 minute.

Nickel33, 2019-01-21:
Leaving open until Mac behavior is verified.

jeff101, 2021-01-04:
Interestingly Python seems to have fixed this issue internally. If I recall, previously the time would be loaded from the OS when the application started, but it wouldn't be updated if the user changed the time manually. It now works as expected on both Windows and Mac.