LowellInstruments/Domino #89: Software doesn't update time if the user changes the computer time

jeff101, 2018-12-19:
The software is not adjusting the "computer time" field when the date and time are adjusted by the user in the control panel. Rather, the user must close and re-open the application.  This is important because many customers set their logger to UTC time but normally have their PC on local time.  So without this change, they need to set the PC time zone before opening the application.  That is unusual workflow.

jeff101, 2019-07-28:
I looked into this issue months ago and the consensus on Stack Overflow was that this is a known issue with the datetime module. Once you imported it, it doesn't register manual clock changes. Now mysteriously it's working! There must have been an update that fixed it.

jeff101, 2019-07-29:
Never mind. It works if you change the computer time, but not if you change the time zone.

jeff101, 2019-07-29:
Now I remember. The solution isn't very elegant and may cause cross platform difficulties.
https://stackoverflow.com/questions/4360981/make-python-respond-to-windows-timezone-changes
If the user changes the time zone, then Domino must be closed and reopened.

Nickel33, 2019-07-30:
Not ideal, but good enough for now.  Closing.

closed 2019-07-30 by Nickel33