LowellInstruments/Domino #95: Change color of the icon to better match the LI logo

Nickel33, 2018-12-20:
The purple icon look okay, but it would be better if it matched the LI icon colors somehow or was black and white (not so it matches the real-life dog, but so it doesn't clash with the LI logo.)

jeff101, 2018-12-21:
Have a look at the icon branch

Nickel33, 2019-01-02:
That looks better.  One more place to change:  In the Info box, change the "!" icon to the Domino icon.

jeff101, 2019-01-30:
[image]

I just changed the info icon to the Domino icon and I don't like it as much. It kind of just looks like a decoration and it isn't clear to me that it's an info button. What do you think?

Nickel33, 2019-07-24:
The "I" icon in 0.5.8 looks good.  Closing issue.

closed 2019-07-24 by Nickel33