LowellInstruments/Domino #96: Logger Status is not obvious enough

Nickel33, 2018-12-20:
The status of the logger is not obvious enough.

There are really two different high-level pieces of information that need to be communicated to the user.
1. The device is connected/is-not-connected.
2. The device is running/delayed/stopped/in-an-error-state.

I propose splitting the two pieces of information up.  The connected or not connected status could be a USB connector and the logger with data.  Something like this:
[image]
It should also contain text.  "Logger is connected to USB."
If a logger is not connected the logo could be:
[image]
with the text: "No Logger Detected"

The running/delayed/stopped/in-an-error-state would have a another set of icons.  The rabbit works for run.  The stop sign works for stop. But icons are needed for error and delayed.  I am open to suggestions.

jeff101, 2019-01-24:
[image]

jeff101, 2019-01-24:
[image]

jeff101, 2019-01-24:
[image]

jeff101, 2019-01-24:
Status will be moved from the the top, as pictured above, to a status bar on the bottom of the window.

Nickel33, 2019-07-24:
Looks good in 0.5.8.  Closing.

closed 2019-07-24 by Nickel33