LowellInstruments/Domino #100: Instructions needed for how to get a non-executable running on a new computer

Nickel33, 2019-01-02:
STORY:
As a new user of the software, I need step-by-step instructions for how to download Domino and get it running on my computer.  What additional software is required?  Where should the files be stored? What changes to the environment variable are required etc.

jeff101, 2019-01-14:
Is this specifically for in-house testing? Or do you propose that an end user may want to run our software via their own python installation, not from a "compiled" exe? If it's just for in house purposes, here are the instructions I put together:
https://docs.google.com/document/d/1XTJbaWQCGlz6biqpjBedaM95eXBzJDoDuvx9tnO_Ktc/edit?usp=sharing

Nickel33, 2019-01-21:
In house is fine for now.

Nickel33, 2019-01-21:
I added the instruction link to the Readme for Domino.

closed 2019-01-21 by Nickel33