LowellInstruments/Domino #102: On Mac, logger drive is not being removed from finder/desktop when logging starts

Nickel33, 2019-01-04:
STORY:
As a user, the MAT_LOGGER drive should not be shown as a storage option after the logger has been started.  It is confusing that the drive shows up in Finder/Desktop but is not usable.  If I try to access the drive, it shows no files and hangs.

Notes:  On a PC, when the logger is starte, the MAT_LOGGER media is no longer accessible and an open file explorer windows close.  But on the Mac, when the logger is started, a open window showing the drive contents stays open and the drive still shows as available in Finder.   Attempting to save a MAT.cfg file to the drive will hang Domino.  Accessing the drive via Finder will produce the "beach-ball" until it eventually times out.  Physically disconnecting the logger from the USB port is the only way to make it disappear.  Upon reconnection, the drive is not shown, until the logger is stopped (as expected.)

This may be a difference between Windows and Mac OSes with respect to how they work with removable media.  On Windows, the MAT_LOGGER drive, after un-enumeration, is still present as "Removable Disk (X:)".  Perhaps the Mac is not recognizing the un-enumeration for the logger.  On USB reconnect, on a PC the drive still shows up as "Removable Disk (X:)" but on the Mac the drive is not visible at all.  The behavior is different when I user a card reader on the MAC.  In that case, when I remove the media (not the reader itself) the Finder recognizes that the media has been removed and I can not access the drive (as expected.)  So there must be something different between how the card reader and logger send the signal that the media is no longer available.

jeff101, 2019-08-08:
Un-assigning myself for now as it's a future feature.

jeff101, 2021-01-04:
Write a several sentence description of this problem for Jim