LowellInstruments/Domino #104: Unclear what is happening when logger connected but port is busy

Nickel33, 2019-01-04:
If the logger is being used by another application OR if the logger is unresponsive on the COM port, Domino can get stuck where it is neither connected or disconnected.  I think that some of these are default values and they need to be changed.  None of the fields should be filled in.  It should not say that the device is running etc.  In addition, the stop and start buttons should BOTH be disabled.

[image]

To re-create, connect with matconfig and then run Domino.

jeff101, 2019-03-06:
I believe this is now outdated

Nickel33, 2019-07-24:
The behavior looks logical and clear in 0.5.8. Closing.

closed 2019-07-24 by Nickel33