LowellInstruments/Domino #108: When attempting to save a mat.cfg file with an invalid setting, pop up a warning box.

Nickel33, 2019-01-14:
STORY:
As a user, I want to know, definitely, that something went wrong when attempting to create a setup file with invalid settings.

Details:
Set a end time in the past.  The field is highlighted in yellow, but if I ignore the error, Domino lets me save the file anyway but the date in the field changes to 1/1/2020 while the date in MAY.cfg  is 1/1/2096.  There are two issues:  1. The file should not have been saved at all, rather a pop-up should be shown. Second, the date and time in the field should not be automatically updated, since chances are that the user only got a hour, month or year wrong, not the whole field.

jeff101, 2019-07-25:
[image]
A related error is that a file size should not be displayed

jeff101, 2019-07-26:
Domino no longer highlights errors in yellow and notifies the user of an error via a popup.

Nickel33, 2019-07-30:
Domino is NOT notifying the user of an error.  Specifically, set an end time in the past and Domino allows the file to be saved without a warning.