LowellInstruments/Domino #123: Add a "secret" connect/disconnect button

jeff101, 2019-01-24:
TODO
As production engineer, I work with multiple software applications that need to talk to the MAT logger when assembling, calibrating, and testing. Domino is "greedy" with the com port, meaning that as soon as a MAT logger is discovered, a connection is opened and other software applications are unable to talk to the logger. If I want to use another application, I have to close Domino so it gives up the port. 

DONE
Add a way for me to close the connection, and have it stay closed, until I open it again, and keep it hidden as to not confuse end users.

jeff101, 2019-02-14:
On Domino-123

Nickel33, 2019-02-17:
Not fixed.  COM port remains open after CNTRL-Click.

jeff101, 2019-03-06:
on branch Domino-123_take_2

jeff101, 2019-03-07:
by the way, see the status bar for the status of the manual connection. It will display either "Auto Connect" or "Manually Disconnected"

Nickel33, 2019-07-24:
Fixed. 0.5.8. Closing.

closed 2019-07-24 by Nickel33