LowellInstruments/Domino #157: Start Time Default Time is 1970, PM time not possible?

Nickel33, 2019-04-21:
Story:
As a user that wants to use a start time, I expect the seed time to be in the future, not the past. Also, I want to be able to set the time in the second half of the day.

Details: 
Select start time and the date is 1970.  This is a pain, as the calendar is essentially useless.  Also, it is not clear if it is possible to set a time in PM.  Selecting a time of 22:00:00 is flagged as invalid, yet there is no AM/PM selection

jeff101, 2019-05-07:
I'm not able to recreate the 1970 issue, however I do see the problem with the time. I'll fix that first.

Nickel33, 2019-07-24:
0.5.8
Time problem is fixed. 
Can't reproduce the 1970 issue.
Closing.

closed 2019-07-24 by Nickel33