LowellInstruments/Domino #159: Installer: Don't load driver on Windows 10, not needed

Nickel33, 2019-04-25:
Story:  As a windows 10 unser, it is not necessary to load the Windows 10 CDC driver.  When the installer attempts to load it, and it fails, I am confused and think something went wrong.

Background:  Windows 10 includes CDC (serial) support natively (as of update??) .  It is not needed.  The driver should be loaded for Windows 7 & 8.  For Windows 7 it should work.  For Windows 8 it will throw an error and they will need to manually load the driver by following the Windows 8 driver loading procedure available on the lowellinstruments.com website under FAQs.

jeff101, 2019-05-06:
I think I have this fixed but I don't have a Windows 10 machine.

Nickel33, 2019-05-13:
Not fixed in Windows 10.

[image]

Nickel33, 2019-07-17:
New message: 
[image]

jeff101, 2019-07-28:
I have this working on my Windows 8 computer but I need to test it on Windows 10. On the second last line of the innosetup script, change 
Filename: "{app}\drivers\dpinst.exe"; OnlyBelowVersion: 0,10
to (remove 0,)
Filename: "{app}\drivers\dpinst.exe"; OnlyBelowVersion: 10

Nickel33, 2019-07-30:
Verified fixed 0.6.0.1.

closed 2019-07-30 by Nickel33