LowellInstruments/Domino #164: lid file crashes Domino when "Convert" is clicked

richardsc, 2019-05-09:
Not sure if it's the file, but when I try and convert Domino crashes. Are there logs somewhere that might indicate what is going on?

[1904108_Test_(3).zip](https://github.com/LowellInstruments/Domino/files/3160145/1904108_Test_.3.zip)

richardsc, 2019-05-13:
Also, this is on Windows 7 (if that might matter).

jeff101, 2019-07-16:
I'm able to convert this file to all possible output types, with and without averaging enabled. Perhaps one of the bug fixes in the past couple months corrected this problem?

richardsc, 2019-07-16:
Same result as before for me :cry: 

I click "Convert" and get as below. Are there any kind of program log files that I could send that might indicate what is happening?

[image]

Nickel33, 2019-07-18:
This is probably the same issue as #169.  Try release 0.5.8, due out later today (July 18th) and see if it works.

jeff101, 2019-07-27:
@richardsc , thanks for reporting this bug. I just took a more careful look and Nick was correct; it has the same cause as #169. It was corrected in 0.5.7, however 0.5.8 is now available on Github (not yet on the Lowell Instruments website). https://github.com/LowellInstruments/Domino/releases

Nickel33, 2019-07-30:
Declaring this issue fixed 0.5.8.

closed 2019-07-30 by Nickel33

richardsc, 2019-08-01:
Can confirm that it appears to work! Thanks!