LowellInstruments/Domino #170: Excessive CPU use on 0.5.6 a 0.5.7

Nickel33, 2019-07-17:
The CPU use is up around ~30% for versions 0.5.6 and 0.5.7.  The program is running very sluggishly.  Three machines tested Windows 7 and 10.

jeff101, 2019-07-18:
Through process of elimination, I have discovered this problem is due to code on the "Start/Stop Device" tab.  Now to figure out what the problem is...

jeff101, 2019-07-18:
Well that was easy. The very first thing I tried fixed it! QThread has a method called sleep. It seems as though a value less than 1 causes the CPU usage to skyrocket. I suspect the function is internally rounding down to the nearest int, in this case 0 seconds, therefore no delay. QThread also has a method called msleep (milisecond sleep). A value of 250 has the CPU usage back under 1%.

Nickel33, 2019-07-18:
Verified fixed in 0.5.8.

closed 2019-07-18 by Nickel33