LowellInstruments/Domino #173: Eliminate Windows SmartScreen Protection

Nickel33, 2019-07-24:
Story:  As a user, when I download and install Domino I don't expect to get a scary warning from Windows that the program is risky and should not be run.

Additional comments:  We probably need a code signing certificate.  Also, the installer should not get any hits when scanned at Virustotal.com

jeff101, 2019-07-25:
Look into code signing certificate. Check out Comodo.

jeff101, 2019-07-29:
Code signing is up and running. The user will still get a SmartScreen notification, but at least now it will be blue and say the software is from Lowell Instruments.

Nickel33, 2019-07-30:
Partially fixed.  At least the warning is not as severe and the publisher is shown as Lowell Instruments LLC.  Closing for now.

closed 2019-07-30 by Nickel33