LowellInstruments/Domino #180: Alert the user that temperature is recommended when using magnetometer

Nickel33, 2019-07-30:
Story:  As a user who is not interested in temperature, I want to disable the temperature channel to avoid using up memory and battery. But I didin't know that the magnetometer is temperature compensated.  I would like to know that before I save the MAT.cfg file.

Implementation:  Add a pop-up that notifies the user that if the magnetometer is enabled, the temperature should be enabled too.

jeff101, 2019-08-07:
The wording was off the cuff... you may want to tweak it.

Nickel33, 2019-08-08:
It appears that the fix created another problem.  Now if there is no error, the file won't save at all.  I can only get the setup file to save with Temperature disabled.