LowellInstruments/Domino #181: The ConverterTable class is confusing

jeff101, 2019-07-30:
ConverterTable should not have any responsibilities such as handling conversions and showing dialog boxes. Make it a simple class for manipulating the converter table.

jeff101, 2019-07-30:
Add a status column
Move "containing folder" all the way to the right
If there are converted and unconverted files, prompt the user to reconvert files, otherwise ignore them
Prompt user to overwrite files; give option to overwrite all remaining files

Nickel33, 2019-08-12:
A few minor changes:
1. Change column headings in the table to left justified to match the values below them.
2. Make column headings bold or underlined or something to differentiate them from the values below.
3. Make the default width of the "containing folder" column ~50% wider to accommodate longer paths names.
4. Make it more obvious that a file has been converted.  The Status change from "unconverted" to "converted" is too subtle.  Maybe make the whole row a different color or font or highlighted etc.

Nickel33, 2019-08-13:
Really close.  The only issue that is not done is 2 from above.  The top row of the column needs something else to make it pop.  Can you put a double-underline? Or a single underline and make it dark grey?  See how the top row blends into the table in these shots below:
[image]

[image]

jeff101, 2019-08-17:
Interesting, it looks like Windows 10 renders things differently. Here it is on Windows 7.
[image]

jeff101, 2019-08-17:
I'll talk to you about this on Monday

Nickel33, 2019-10-23:
Looking good. Closing.

closed 2019-10-23 by Nickel33