LowellInstruments/Domino #198: Setting time in Windows 10 causes unresponsive behavior

Nickel33, 2019-09-23:
Story:
A user who lives in a timezone east of GMT starts Domino and connects to the logger.  They want to set the time to GMT and only remember then that their computer clock is set to local time (GMT + x).  So they open Windows Date & Time, uncheck "Set Time Automatically" and manually set the time back one hour.  When they look in Domino, the "Computer Time" is updating normally, but the logger time and real time values are not updating.

Additional Info: If the starting time is in the past and the time is adjusted forwards (later) it works. Also, if the time zone is updated in either direction, Domino does not show a change in the "Computer Time" and thus this is not a problem either.

Done:  Domino does not become unresponsive when the clock is set in the past relative to the logger time.

jeff101, 2019-09-24:
Laying in bed last night, I realized why this happens. The "start/stop device" tab has a scheduler that keeps track of the various commands (get time, get sensors, etc) and at what time they next fire. By setting the time back, the commands don't get dispatched. If you waited an hour everything would probably start working again. 

Possible fixes: 
Monitor the delta between computer time and the command queue. If it is > than 30 seconds, reset the times. 
Use a timer to generate a count every x-hundred ms. Base the queue on counts, not computer time.

jeff101, 2019-09-24:
I have confirmed this hypothesis by settings my clock back one minute. After 1 minute elapses, the connection begins working as usual.

Nickel33, 2019-10-23:
This issue is fixed.  Jeff, how did you fix it?  Please respond and close it.

jeff101, 2019-10-28:
I solved this using the time.monotonic() function

> Return the value (in fractional seconds) of a monotonic clock, i.e. a clock that cannot go backwards. The clock is not affected by system clock updates. The reference point of the returned value is undefined, so that only the difference between the results of consecutive calls is valid.

I was able to use the existing code but replace time.time() calls with time.monotonic()

closed 2019-10-28 by jeff101