LowellInstruments/Domino #199: Crash when in status screen for an hour or two.

Nickel33, 2019-09-24:
0.6.2 RC4.

There are occasional crashes when the logger is connected to the status screen for an extended period of time. The crash occurs for me every hour or two.  This is the traceback:

Traceback (most recent call last):
  File "gui\start_stop.py", line 82, in query_slot
  File "gui\start_stop_updater.py", line 84, in command_handler
  File "gui\start_stop_updater.py", line 111, in update
  File "_strptime.py", line 565, in _strptime_datetime
  File "_strptime.py", line 362, in _strptime
ValueError: time data '2019/09/23 17:\n\rGTM' does not match format '%Y/%m/%d %H:%M:%S'

jeff101, 2019-10-01:
I upped the delay between commands to see if it will fix the problem.

jeff101, 2019-10-02:
I caught this on the port monitor today. Interestingly the packet had the correct length so it was able to sneak past that check. I put a try block in the code where the time conversion takes place so it won't cause a crash next time.

Nickel33, 2019-10-23:
I haven't seen this in a while using 0.6.2.  Closing.

closed 2019-10-23 by Nickel33