LowellInstruments/Domino #220: No battery/bad battery: confusing message

Nickel33, 2019-11-22:
0.6.3

When the logger's battery is missing the message is confusing.  "Device is running - Safe Shutdown"
[image]

This is not accurate. In this more the logger returns a status of 0x10.  This indicates the logger has stopped recording because the battery is bad.  In MAT Logger Commander the message is better, but still wrong:
[image]

What the message should read is: Bad battery, replace now.

Nickel33, 2019-11-22:
Consider adding a new "error" icon in place of the rabbit and hand symbol.  Or just use the hand symbol for this error state.

Nickel33, 2019-11-22:
There is a similar issue when there is a MAT.cfg error.  The logger is running but stopped at the same time:.
[image]

Nickel33, 2019-11-26:
Four Types of Errors:
Error: Bad Battery Replace Now
Error: SD Card Problem
Error: Invalid Setup File
Error: Device Error 0xXX

Use an triangular icon in place of the rabbit and/or hand icon.

jeff101, 2020-01-22:
I've discovered a couple more minor firmware bugs (with the new firmware).

To recreate them:
First bug - start with a device containing a valid MAT.cfg file. Open the SD and delete the file. Within 5 seconds Domino will report there isn't a valid setup file. Now add the setup file back. The channels will update to whether they are enabled or not, but the error message will remain. I've confirmed this is a firmware issue by issuing an STS in Tera Term (code 0x09).

Second. For this one first remove the battery and connect to USB. Again, start with a valid MAT.cfg file. Domino will report the battery is critically low (STS code 0X11 as expected). Now delete the MAT.cfg file and Domino will report the two errors (STS code 0x19). Now click start, and all mention of the low battery has disappeared. (STS code 0x09). The battery bit doesn't get set again until the MAT.cfg file is restored, and start is clicked.

Nickel33, 2020-02-06:
The battery issue is not really a firmware problem.  It is by design.  The only 2 times the battery voltage is checked is:
1. On Powerup
2. Every time a 32K page is written to the SD card.
The symptoms you are describing are consistent with that behavior.  We can't fix the battery error.  We can suppress the MAT.cfg error.  I will write a new bug for how to do it.

closed 2020-02-06 by Nickel33