LowellInstruments/Domino #221: POSIX Time Stamp is not accounting for time zone

Nickel33, 2019-11-25:
0.6.3
The POSIX time output option is not accounting for the time zone of the clock that was used when the device clock was set.  As a result, when converting POSIX timestamps to human dates, the time is off by the number of hours from GMT that the reference clock was set.

To put it another way, POSIX time assumes that the logger was set to GMT when started.  Note the time, set the loggers clock,  start it, stop it, convert the file using POSIX time output, copy the starting time 

This is slightly confusing.

Since the time zone is not stored in the logger, there is no way to grantee know what the correct time is.  So this should probably be handled with an explanation in the software and documentation.

Nickel33, 2019-11-26:
Change to: Local POSIX Time Stamp (seconds since Jan 1, 1970)

jeff101, 2020-01-22:
Upon further investigation, there are a couple issues here that need to be sorted out. As usual, time zones and daylight savings time are a big pain. It's true that the converted POSIX time will be local time assuming the .lid conversion is done on a computer running the same timezone and DST as the logger when it was started. We'll need to think about this some more...