LowellInstruments/Domino #225: Driver installation induced hard crash Windows 7

Nickel33, 2019-11-25:
A crash can result if Domino is open when a logger is connected for the very first time and the driver must be installed in Windows 7.

Here is how it happened:

1. Connected a logger to COM 7.
2. Opened device manager and uninstalled the driver (and deleted it).
3. Opened Domino.
4. Reconnected a logger to COM 7.
5. Windows started the installation and Domino crashed.

This is a fairly specific bug on a dead OS. So I don't think it requires fixing by itself.  However, it does point to a unhandled exception that should be looked at in case it is more general.

jeff101, 2020-01-13:
Can we close this ?

Nickel33, 2020-02-06:
Closing.  Windows 7 is no longer supported by us or Microsoft.

closed 2020-02-06 by Nickel33