LowellInstruments/Domino #230: Suppress MAT.cfg file errors

Nickel33, 2020-02-06:
The MAT.cfg file error is almost always a red herring.  It usually means that the user just has not put the MAT.cfg file on the SD card yet, and because of the firmware behavior, it can't be cleared without starting the logger.  But this is very counter-intuitive:  who would start a logger that says that it is in an error state?

The only solution I can come up, short of magically updating the firmware in all the field devices, is to suppress the error.

Nickel33, 2020-02-06:
To clarify, just suppress the message on the status screen.  Keep this pop-up:
[image]

jeff101, 2020-02-24:
Error messages will no longer be displayed on the status screen. From now on, when the "Start Running" button is pressed, Domino will query the logger (STS) and report any errors in a dialog as above.

Nickel33, 2022-12-08:
Verified fixed. Closing.

closed 2022-12-08 by Nickel33