LowellInstruments/Domino #254: Don't allow two instances of Domino to be open at the same time

Nickel33, 2021-03-11:
Story:  User "Lisa" is a Mac person but is using a Windows PC because Domino is not fully supported on a Mac.  When opening Domino, she accidentally opened Domino twice.  Then when she connected the logger, it appeared that the logger was not connecting.  In reality, it was connecting, but to the other open version of Domino.  (Since they were directly on top of one another, it wasn't immediately obvious that there were two copies running.) This was confusing and she called technical support.

To do: Check if Domino is running before opening a second copy.  If it is already running, bring Domino to the front.

jeff101, 2021-05-06:
https://stackoverflow.com/questions/380870/make-sure-only-a-single-instance-of-a-program-is-running

jeff101, 2022-07-28:
Solved using Socket_Singleton https://pypi.org/project/Socket-Singleton/#description
Unclear if there will be any issues with the socket getting stuck open and Domino refusing to launch.

Nickel33, 2022-09-27:
Partial Fix.  If an old version is running, it doesn't detect that the old version is open.  Can you use a more generic search for Domino?

jeff101, 2022-10-01:
This is not going to be easy. While the the current version will not detect older versions of Domino, it will work with future versions. Isn't it unlikely we need to worry about a user having two versions open at once? I can pursue a fix if you want.