LowellInstruments/Domino #257: Notify user when A/D values are zero for enabled channels.  Suggest reset.

Nickel33, 2021-05-23:
The magnetometer is sometimes getting stuck and reports all zero (pre-compensated) values.  This is confusing because the logger otherwise seems to be working and the compensated values are non-zero.  So, we should notify the user when the magnetometer binary values are zero and suggest a software reset.  It is a clunky workaround for a firmware bug, but better clunky than nothing.

jeff101, 2021-11-18:
If the magnetometer reports zeros for the raw values three readings in a row, then indicate that there is a possible problem with the logger and we recommend a reset. Give them the option to reset or dismiss.

Nickel33, 2022-09-28:
Looks good.  Closing.

closed 2022-09-28 by Nickel33