LowellInstruments/Domino #258: Attempt to prevent customer from converting lid files while connected via USB

jeff101, 2021-06-25:
STORY
The MAT-1 USB connection is not a high speed connection and it is not sufficient to convert a data file. Not only is the conversion extremely slow, but there is reason to believe it can crash the device. 

DONE
Ideally, definitively determine if the file on the MAT-1 and raise a warning
If that's not possible, determine if the file is in the root directory and raise a warning that the file MAY be on a device.

jeff101, 2021-11-18:
If the file is on your C drive then no warning. If the file is on a removable disk, pop a warning that it's recommended that you copy the file from the memory card to your local disk before converting.

Nickel33, 2022-05-24:
#258 is duplicate.  Closing.

closed 2022-05-24 by Nickel33