LowellInstruments/Domino #260: Add "Trap Orientation" as a file output option

Nickel33, 2021-07-07:
STORY:  The user has mounted a MAT-1 Logger inside a lobster pot. They want to know if the trap was vertical during the duration of the deployment.  The existing output options are not appropriate.  

DONE: Add an output option that shows the orientation of the logger in terms of degrees from vertical.  Assume the logger is mounted vertically in the trap with the end cap UP. The output should be such that if the trap is perfectly vertical the result will be 0.  If it is on its side the output will be 90.  If it is upside down it will read 180.

Additional Info: The magnetometer channels will not be used for this measurement.  The logger will not be oriented about the vertical axis, so we won't be able to differentiate between a trap on its side or on its end. This feature has been requested by NMFS Woods Hole as part of the ropeless fishing project.

Nickel33, 2021-08-18:
What do we output of the magnetometer IS enabled?  Ignore it and give a magnetometer only channel(s).

jeff101, 2021-08-23:
> What do we output of the magnetometer IS enabled? Ignore it and give a magnetometer only channels.

If temperature and/or magnetometer are enabled, they will be extracted to their own files

Nickel33, 2021-09-16:
Done.

closed 2021-09-16 by Nickel33