LowellInstruments/Domino #268: Add new MAT Library with support for DOT Logger Water Detection sensor

Nickel33, 2022-05-20:
The DOT logger has a water detection sensor.  It has been added to the MAT library, but the current version of Domino uses an older version of the library.  Request to switch to the new MAT library so that water detection works and DOT .lid files parse correctly. Sample file attached.
[3333333_osu_20220511_154733.zip](https://github.com/LowellInstruments/Domino/files/8741258/3333333_osu_20220511_154733.zip)

jeff101, 2022-06-10:
The sample file now converts

Nickel33, 2022-06-13:
The file converts, but it creates a second file.  This is a fault of the MAT Library.  The DO, Temperature and Water detection sensors all record at the same rate, so there should only be one file.  I will close this bug and create a new one for the file conversion.